bandasaikrishna/Autonomous_Mobile_Robot #4: Issues in case of controller_check.launch

vijaybande, 2020-12-28:
while running controller_check.launch I am getting 
" Device File not available. Aborting read"

yuki-ubuntu, 2021-03-19:
I am also suffering from the same error.

```
... logging to /home/user/.ros/log/24fe7864-888a-11eb-bda4-bb6933ebd3a4/roslaunch-user-3515.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user:39347/

SUMMARY
========

PARAMETERS
 * /joint_limits/left_wheel_joint/has_acceleration_limits: False
 * /joint_limits/left_wheel_joint/has_effort_limits: False
 * /joint_limits/left_wheel_joint/has_jerk_limits: False
 * /joint_limits/left_wheel_joint/has_position_limits: False
 * /joint_limits/left_wheel_joint/has_velocity_limits: True
 * /joint_limits/left_wheel_joint/max_acceleration: 0.0
 * /joint_limits/left_wheel_joint/max_effort: 0
 * /joint_limits/left_wheel_joint/max_jerk: 0
 * /joint_limits/left_wheel_joint/max_position: 0
 * /joint_limits/left_wheel_joint/max_velocity: 2.62
 * /joint_limits/left_wheel_joint/min_effort: 0
 * /joint_limits/left_wheel_joint/min_position: 0
 * /joint_limits/left_wheel_joint/min_velocity: -2.62
 * /joint_limits/right_wheel_joint/has_acceleration_limits: False
 * /joint_limits/right_wheel_joint/has_effort_limits: False
 * /joint_limits/right_wheel_joint/has_jerk_limits: False
 * /joint_limits/right_wheel_joint/has_position_limits: False
 * /joint_limits/right_wheel_joint/has_velocity_limits: True
 * /joint_limits/right_wheel_joint/max_acceleration: 0.0
 * /joint_limits/right_wheel_joint/max_effort: 0
 * /joint_limits/right_wheel_joint/max_jerk: 0
 * /joint_limits/right_wheel_joint/max_position: 0
 * /joint_limits/right_wheel_joint/max_velocity: 2.62
 * /joint_limits/right_wheel_joint/min_effort: 0
 * /joint_limits/right_wheel_joint/min_position: 0
 * /joint_limits/right_wheel_joint/min_velocity: -2.62
 * /mobile_robot/joints_update/publish_rate: 50
 * /mobile_robot/joints_update/type: joint_state_contr...
 * /mobile_robot/mobile_base_controller/angular/z/has_acceleration_limits: True
 * /mobile_robot/mobile_base_controller/angular/z/has_jerk_limits: True
 * /mobile_robot/mobile_base_controller/angular/z/has_velocity_limits: True
 * /mobile_robot/mobile_base_controller/angular/z/max_acceleration: 0.5
 * /mobile_robot/mobile_base_controller/angular/z/max_jerk: 1
 * /mobile_robot/mobile_base_controller/angular/z/max_velocity: 1.0472
 * /mobile_robot/mobile_base_controller/base_frame_id: base_link
 * /mobile_robot/mobile_base_controller/cmd_vel_timeout: 0.25
 * /mobile_robot/mobile_base_controller/left_wheel: left_wheel_joint
 * /mobile_robot/mobile_base_controller/linear/x/has_acceleration_limits: True
 * /mobile_robot/mobile_base_controller/linear/x/has_jerk_limits: False
 * /mobile_robot/mobile_base_controller/linear/x/has_velocity_limits: True
 * /mobile_robot/mobile_base_controller/linear/x/max_acceleration: 0.025
 * /mobile_robot/mobile_base_controller/linear/x/max_jerk: 2.0
 * /mobile_robot/mobile_base_controller/linear/x/max_velocity: 0.1
 * /mobile_robot/mobile_base_controller/linear/x/min_acceleration: -0.025
 * /mobile_robot/mobile_base_controller/linear/x/min_velocity: -0.1
 * /mobile_robot/mobile_base_controller/pose_covariance_diagonal: [0.001, 0.001, 10...
 * /mobile_robot/mobile_base_controller/publish_rate: 50.0
 * /mobile_robot/mobile_base_controller/right_wheel: right_wheel_joint
 * /mobile_robot/mobile_base_controller/twist_covariance_diagonal: [0.001, 0.001, 10...
 * /mobile_robot/mobile_base_controller/type: diff_drive_contro...
 * /mobile_robot/mobile_base_controller/wheel_radius: 0.04
 * /mobile_robot/mobile_base_controller/wheel_radius_multiplier: 1.0
 * /mobile_robot/mobile_base_controller/wheel_separation: 0.18
 * /mobile_robot/mobile_base_controller/wheel_separation_multiplier: 1.0
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
[... 56 lines cut ...]
[ERROR] [1616141252.427424013]: Device File not available. Aborting read
[ INFO] [1616141252.430225838]: Odometry params : wheel separation 0.18, left wheel radius 0.04, right wheel radius 0.04
[ INFO] [1616141252.437641994]: Adding left wheel with joint name: left_wheel_joint and right wheel with joint name: right_wheel_joint
[ INFO] [1616141252.477879388]: Dynamic Reconfigure:
DynamicParams:
	Odometry parameters:
		left wheel radius multiplier: 1
		right wheel radius multiplier: 1
		wheel separation multiplier: 1
	Publication parameters:
		Publish executed velocity command: disabled
		Publication rate: 50
		Publish frame odom on tf: enabled
[ERROR] [1616141252.527265299]: Device File not available. Aborting read
[ERROR] [1616141252.527408167]: Device File not available. Aborting read
[INFO] [1616141252.528493]: Controller Spawner: Loaded controllers: /mobile_robot/joints_update, /mobile_robot/mobile_base_controller
[ERROR] [1616141252.627321803]: Device File not available. Aborting read
[ERROR] [1616141252.627481930]: Device File not available. Aborting read
[INFO] [1616141252.628878]: Started controllers: /mobile_robot/joints_update, /mobile_robot/mobile_base_controller
[ERROR] [1616141252.727312585]: Device File not available. Aborting read
[ERROR] [1616141252.727488953]: Device File not available. Aborting read

```

If you have a solution, I would appreciate it if you could let me know.

vijaybande, 2021-04-08:
this issues resolved

closed 2021-04-08 by vijaybande

mehmetaksu2034, 2021-05-20:
Hi, how can you solve this problem? Can you help me pls.

vijaybande, 2021-05-21:
Kindly check your i2c buses are available or not??