bandasaikrishna/Autonomous_Mobile_Robot #5: Remote I/O error. Failed to write to I2C Slave 0x8 @ [write_byte():write 121], Could not read from I2C slave 0x8, [read_byte():read 121]

vijaybande, 2021-04-08:
I am gettig below errros while running controller_check.launch. i2c slaves 0x08 and 0x09 were detected on Rpi

[ERROR] [1617861674.244743164]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.252197838]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.252879451]: Remote I/O error. Failed to write to I2C Slave 0x8 @ [write_byte():write 121]
[ERROR] [1617861674.253275023]: Remote I/O error. Failed to write to I2C Slave 0x9 @ [write_byte():write 121]
[ERROR] [1617861674.344301937]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.345627664]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.444579302]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.446186018]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.544354689]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.546916455]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.644397524]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.645817938]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.744349213]: Could not read from I2C slave 0x8, [read_byte():read 121]

vijaybande, 2021-04-20:
Resolved.

closed 2021-04-20 by vijaybande

audricsaragih, 2021-05-13:
Could you tell me please how you solved this?

vijaybande, 2021-05-13:
1.) Make sure that the both i2c are enabled on the board i.e i2c-0 and i2c-
1
2.) Connect the arduino slave 08 to i2c-0 and arduino slave 09 to i2c-1

But may I know how it is working in the link given below ??
https://youtu.be/Uz_i_sjVhIM

audricsaragih, 2021-05-13:
Yes I've done that both.
But when i run :

> i2cdetect -y 0

or

> i2cdetect -y 1

It didn't catch any adress
I'm stuck there too

vijaybande, 2021-05-13:
Are u using raspberry pi or Jetson nano??

vijaybande, 2021-05-13:
Can we connect over the call??

audricsaragih, 2021-05-13:
I'm using Raspberry Pi.
Sure, you can email me first at audric.saragih@gmail.com

robotos2023, 2022-12-07:
I got the same error. I am using Raspberry pi 4 model b. Can you please tell me how you do it? I'd appreciate it a lot!

jomar04, 2023-05-25:
Can I ask how you resolve it?

Rithikrkm, 2025-03-10:
Bro Im having the same issue and I've been stuck here for a while, can you pls share how u resolved this issue