bandasaikrishna/Autonomous_Mobile_Robot #9: I am gettig below errros while running controller_check.launch. i2c slaves 0x08 and 0x09 were detected on Rpi and Im using raspberry pi 4

gndeepan, 2022-06-13:
[ERROR] [1617861674.244743164]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.252197838]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.252879451]: Remote I/O error. Failed to write to I2C Slave 0x8 @ [write_byte():write 121]
[ERROR] [1617861674.253275023]: Remote I/O error. Failed to write to I2C Slave 0x9 @ [write_byte():write 121]
[ERROR] [1617861674.344301937]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.345627664]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.444579302]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.446186018]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.544354689]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.546916455]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.644397524]: Could not read from I2C slave 0x8, [read_byte():read 121]
[ERROR] [1617861674.645817938]: Could not read from I2C slave 0x9, [read_byte():read 121]
[ERROR] [1617861674.744349213]: Could not read from I2C slave 0x8, [read_byte():read 121]

gndeepan, 2022-06-14:
I solved this error but rviz robot model not moving but motors running perfectly not moving 3d model any one plz solve this error

315haoge, 2022-12-01:
> I solved this error but rviz robot model not moving but motors running perfectly not moving 3d model any one plz solve this error

Hi, could you please help me how to solve this problem? Thank you.

robotos2023, 2022-12-07:
Same issues. Would you please help us to solve this problem. Thank yoi!

Rithikrkm, 2025-03-10:
[ERROR] [1617861674.252197838]: Could not read from I2C slave 0x9, [read_byte():read 121]

Im stuck with the above issue, would you please share on how u fixed this prblm